RaceFlightOne/RaceFlightOne #214: wrong motor orientation image when select "invert prop"

robomaniac, 2017-10-23:
this is small detail but the image of the motor orientation image should be changed when you selected the "invert prop" in setup.

When you do the motor wizard calibration, the prop in the image spin wrong direction.

Arti3DPlayer, 2017-10-23:
@robomaniac In what wizard and on what step ?

Arti3DPlayer, 2017-10-23:
Prop direction images are basing on this:
[image]

robomaniac, 2017-10-23:
both wizard and step

let me take you print screen when I get back home

this mean the image are suppose to change, right?
When I have a new build, I go through the wizard and click that little box
you showed in the wizard, then I calibrate my motor and it's the wrong
image.

Arti3DPlayer, 2017-10-23:
@robomaniac Wizards could change this value. So I recommend to do all wizards 1-6 and then change configuration on "configuration" or pid tuning tabs

robomaniac, 2017-10-25:
oh sorry, I just did a "fix motor direction and idle" test and yes the image is showing the right direction.

humm.... I was sure I saw that, maybe it was previous GUI, in the wizard, the very first time you set the quad.

ok please close thread. If I ever see it again, I will come back here :)

thanks!

closed 2017-10-25 by Arti3DPlayer