RaceFlightOne/RaceFlightOne #216: Feature Request for Stick Arm

fubar1179, 2017-11-02:
I have a buddy who is confined to a wheelchair and has partially paralyzed hands. He has custom gimbal stick ends printed that he sticks his thumbs through to be able to fly. He just got his first raceflight build and has learned that there is no way to setup arming from the sticks. Is there anyway that you can add arming from the sticks as at least a CLI option for people who do have disabilities? I understand arming from a switch is safest, but for people like my buddy, stick arm is the safest for him and really the only option.

fubar1179, 2017-11-12:
[image]
This is my buddy.

2mod, 2017-11-20:
hey m8, sry it took me a while - try the following:
assuming u want to put both sticks down and inwards

1. create 4 logical switches ie L1-L4, which basically trigger when the according stick is in the desired position:
ie. a<x / Thr (throttle) / -99 (u can also play around with this value, depending on how sensitive u want it)
or if the stick position is in the positive range:
a>x / Ail (yaw) / 99
and do that for all 4 stick positions (each on an individual logical switch)
u can test if the logical switch is triggering, by moving the stick and verifying ie L1 becoming bold

2. once u have all sticks triggering at the desired positon, create another 2 (ie L5-L6):
a>x / L1 / 0 / L2
a>x / L3 / 0 / L4
if u position the sticks in the desired arm position, both should light up (bold)

3. now to make things easier, create a final logical switch L7 with:
a>x / L5 / 0 / L11 / --- / 3
where "3" defines the number of seconds the sticks have to be in the desired position, before the logical switch triggers (adjust up/down as desired)

now here comes the tricky bit, to which there is probably a more elegant solution, but this works too...

4. create a special function to set the global variable (ie GV1) to 1 (aka arm):
L7 / adjustGV / 1 / (and tick the checkbox to activate the function)

5. create another logical switch (L8) to trigger based on the global variable:
a=x / G1 / 1

these last 2 steps (4&5) will make triggering L7 stick, so it doesnt disarm again after letting go of the sticks

6. create a further special function to override the arm switch with the desired value:
L8 / override / (ur arm channel) / -100/100 (depending on how u setup the arm channel)

the thing is, now ull need the same procedures as above, but with a different stick combination, to disarm
so u gotta repeat 1-5, or at least 2-5, but this time set GV1 to 0

and optionally:
- u could also setup the fc to single arm, so u dont have to wiggle the sticks 2ce
- and u dont have to assign arm to an actual switch in the mixer, since the above works with the channel its self, not the switch

greetz flappy

fubar1179, 2017-11-20:
And I though it was going to be difficult. :P thank you and we'll give this a shot. Would be so much easier if it was a hidden feature to special circumstances like this.

2mod, 2017-11-20:
let me know on slack if ur having trouble