RaceFlightOne/RaceFlightOne #221: FrSky telemetry freezes shortly after takeoff- Builds after 320 (reopening issue #210)

kenkamm, 2017-11-23:
This is reopening issue #210. Proggod asked us to test build 402 to see if it was fixed, but the issue persists. Tested 320 and 402 back to back on the same quad, and 320 works perfectly well, while 402 has intermittent telemetry, occasionally stopping completely. Returning to firmware 320 restores perfect operation of telemetry.

Copied from previous issue #210:
FrSky telemetry doesn’t work reliably. Tested on XSR and R-XSR, both of which running latest FrSky firmware. Telemetry works at first, but after taking to the air the numbers freeze and updates to the Taranis stop. Thus, any voltage or mAh warnings set up in the Taranis don’t work and you end up over-discharging your batteries.

kenkamm, 2017-11-26:
I have lots more flights under my belt now and finding that telemetry is intermittent on 320 as well, where I once thought it was perfect. I get warnings from the taranis about once every couple of flights, not too bad but still not working 100% of the time. I get no such warnings on my betaflight machines. The 320 firmware still seems to work overall better for me than newer builds. Even the very latest, 409, 411, still exhibit dropped telemetry intermittently during a flight.

proggod, 2017-11-28:
414 this should be fixed, i put it in the test-monkeys channel

kenkamm, 2017-11-29:
Thanks! Is it OK to try the very latest build as of now, 417? Or does the fix only apply to 414 specifically?

PGIII, 2017-11-29:
417 is fine

kenkamm, 2017-12-02:
Seems to be working now. Thanks!!!!

justingoesfpv, 2017-12-06:
Initially I had the issue where after arming, telemetry would not work. I also realized that I had a RC of 2.2, since updating to 2.2.1RC1, I've found that telemetry does not work while not armed(you can see hte icon flicker like it wants too on the taranis), but as soon as I arm, i get plenty of data and it seems to stay.

PGIII, 2017-12-12:
everything thing seem fixed now ? will close issue if so

kenkamm, 2017-12-12:
Yes, seems to be working for me, thanks!

closed 2017-12-18 by PGIII