RaceFlightOne/RaceFlightOne #222: GUI Feature request - Wizard status display

2mod, 2017-11-24:
it seems like people still have the problem that they cant arm, because they havent run through all of the wizards, which is usually related to one of the "calibrated" parameters being at 0

wouldnt it be nifty displaying the status of these flags without having to look into the dump? ie by:
- displaying a green tick next or on the wizard buttons
- coloring the buttons red or green
- or something like red/green "led's" next to the buttons or in a separate list

Arti3DPlayer, 2017-11-24:
@2mod it is weird why they didn't run through all wizards...

2mod, 2017-11-24:
@Arti3DPlayer it seems they do, but it can either fail updating the calibration parameter, or they dont realize it failed, probably some also update the fw and assume they dont have to recalibrate at all or only certain things, also assuming it keeps the calibrations from before flashing

Arti3DPlayer, 2017-11-28:
@2mod I redid wizards layout to make them more informative, so maybe this will help. But I will not highlight them. This can cause confusing for people.

closed 2017-11-28 by Arti3DPlayer