RaceFlightOne/RaceFlightOne #226: double arm

gizmatron, 2017-11-28:
always double arm STILL doesn't work.. 409 and  double arm after battery is plugged in no problems.. disarm and rearm and its single arm only .. changing arm mode from 1 to 0 either via the configuration page switch OR via console makes no diference. still double then single NOT always double as it should be

PGIII, 2017-11-29:
ive tested it on frsky, ibus, spektrum, and crossfire and they all double arm all the time, what uart do you have the receiver soldered to

gizmatron, 2017-11-30:
as in said via email its the smart audio.. turn that off double always comes back.. turn it on and lose it again.. while smart audio is on it works via console to change band/channel/power etc all works

kenkamm, 2017-12-02:
I also use smart audio and have this problem as well.

SkeemoFPV, 2017-12-04:
Same here, smart audio and I always get double-single

gizmatron, 2017-12-04:
I'm glad I'm not the only one.. proggod and preston are working on it apparently. for now I set my channel/band/power in console then turn the smart audio off to fly so i keep the double arm.. haven't tried this with the android app yet but hoping it will work too so i  can change freq at the field without the laptop

PGIII, 2017-12-12:
do you still have the issue if the vtx is disconnected but smart audio is still enabled ?

gizmatron, 2017-12-12:
dunno.. not sure I want to unsolder the wire to find out although i suppose its on the edge of the revolt and acesssible.. i'm busy this week so if someone else can test that first ..lol

proggod, 2018-03-19:
This should be fixed, let us know if not.

closed 2018-03-19 by proggod