RaceFlightOne/RaceFlightOne #233: Taranis telemetry ‘FUEL’ wrong

SkeemoFPV, 2017-12-04:
On 419 the mah used reported, is completely wrong. It was previously close, but now getting 35mah reported for a full pack being used.

PGIII, 2017-12-04:
What does it show when it starts ?

2dogrc, 2017-12-04:
Shows 0 on my Taranis on 419, with Bolt.

justingoesfpv, 2017-12-06:
Should show a % of battery used not total mah consumed, there is a variable for battery size which is currently set to 1300 by default. Doesn't change the fact it's incorrect. But doesn't look like current is correct either for the bolt with the default adc(which is supposed to be for the bolt).

2dogrc, 2017-12-10:
Heard there is  beta floating around in the testers section, were you able to fix?

PGIII, 2017-12-12:
Should be fixed in 420, reopen this issue if the problem still persists,  checked with my bolt and it now seems to be pretty close, it is setup to use percent for fuel

closed 2017-12-12 by PGIII