RaceFlightOne/RaceFlightOne #237: Faile Safe issues

theshadowrider, 2017-12-06:
Sorry I don't know how to code and hopefully this is the place to post this.  Is there anyway to be able to make my failsafe, so if I loose signal throttles don't cut and when signal is re-acquired I don't have to re-arm?  Here is the senerio: I have been flying long range (over a mile) and sometimes I momentarily loose signal.  Quad falls to the ground.  I want to be able to have it continue to fly until signal is re-acquired.

PGIII, 2017-12-12:
The issue is that option would be very unsafe for 99% of people, but what radio are you using? crossfire can be setup to hold position instead of cut

theshadowrider, 2017-12-12:
I use SpektrumDX18.  I understand, I was just curious if there is a way to modify mine?

PGIII, 2017-12-14:
https://www.manualslib.com/manual/789010/Spektrum-Dx18.html?page=10 sounds like there is a way to program the failsafe

theshadowrider, 2017-12-14:
The rx I have is SPM4648 do not have a failsafe capabilities.  It would have to be done at the FC.  This is from the manual of the rx.

failsafe
The SPM4648 does not have failsafe capabilities, failsafe should be detected at 
the flight controller.

closed 2018-03-19 by PGIII