RaceFlightOne/RaceFlightOne #238: Switch takes ages to light up in config & Sensor lost

techtechwill, 2017-12-07:
When in the config and taranis is turned on I get 'sensor lost' sometimes. Not sure why.

When I assign my switches, some of the switches (on the left side maybe) seem to take ages to light up green (what you do to confirm the switch is working on that mode as expected etc). Takes 4-5 seconds for the status to turn green after flicking switch.

Loading times need improving when selecting pages and saving (although saving isnt too bad).

Sometimes when I click on the config page it just loads forever and never responds. Have to relaunch.

Current read seems weird. Have it hooked up to my Matek PDB which has a current reader, 3.3V going to current pad on revolt, yet not really getting anything useful on the Taranis.

Still getting a lot of propwash on 419 on my 570g freestyle Alien.

Everything else is golden and flies like a dream, just nitpicking to help with the version 1 :)

PGIII, 2017-12-12:
sounds like you maybe loosing packets which is why the switchs are slow, you may just be too close to the receiver, sensor lost is normal in some wizards since it disables telem when doing some things

PGIII, 2017-12-12:
and try the rc12 tune if you havent

techtechwill, 2017-12-15:
420 has pretty much solved my tuning issues! Only thing that remains is that switches still take a while to recognise in the config, even with taranis the other side of the room. Would love to see less loading between pages too :)

closed 2018-03-19 by proggod