RaceFlightOne/RaceFlightOne #239: Level mode issue

shawreid05, 2017-12-07:
In level mode when quad touches ground to land quad jumps up and does a flip and lands upside down

PGIII, 2017-12-12:
we can look into it, what version are you using ? and does it always do this ?

shawreid05, 2017-12-12:
I am currently running 419, yes always does this amazing flip in level mode
soon as it touches the ground. does not do this in accro.

thanks
Shawn reid

PGIII, 2017-12-12:
does acrolevel have the same issue ? We can try to see if we can replicate it

shawreid05, 2017-12-12:
Yes acro level same thing

Shawn reid

PGIII, 2017-12-26:
Cant get it to do the same thing for me, have you tried the latest version it could be something like an esc desyncing

shawreid05, 2017-12-26:
It’s 419. I will try to put 421 on tomorrow and see if it’s still there. 

Shawn reid

PGIII, 2018-02-26:
im closing this issues since we havent had any issues with level mode in the latest versions, if your problem persists feel free to pm us on FB or slack and we can help you figure it out

closed 2018-02-26 by PGIII