RaceFlightOne/RaceFlightOne #241: Configurator timeouts 324

khandieyea, 2017-12-09:
Heya

I have a v1 revolt, flashed to 324.0, and configurator 324.0 Windows 64 (also Linux 64).

Configurator detects board fine, can read configuratiom from the board, but consistently hits timeout when trying to save/apply changes.

Can set things in console, save, reboot, dump and view the changes.
Have also reflashed in dfu with no noticeably change.

Have tried both from Linux and Windows, Windows configurator is a fresh install ie laptop has never had rf installed prior.

khandieyea, 2017-12-09:
[image]

Arti3DPlayer, 2017-12-12:
@khandieyea check you usb cable also, because you can lost packages while transfer. Also this may happen on very slow pc

PGIII, 2017-12-12:
sounds like a verison mismatch, make sure youre using the 324 firmware

khandieyea, 2017-12-13:
Hey @Arti3DPlayer, sorry I didn't get back to you on slack. and hi @PGIII 

To rule out a few things
- Have tried approx 6 usb cables, varying makes/brands. 
  - These cables work fine in other FC's I have, however I don't any other RF boards, so not 100% ruled out I guess.
- PC's being used are late model laptops (i7, 16gb ram etc), while I'm sure not the fastest machines in the land I'm certain they are fast enough ;)
- I am certain I'm using 0.324.0 for FW and Client (will double check later and post screens)

Cheers

Arti3DPlayer, 2017-12-13:
@khandieyea Sorry, I don't have any idea then what can cause this. Your characteristics seems fine. Could you try on another pc ?

PGIII, 2017-12-28:
i would try the latest version on raceflight.com/beta and see if that fixes your issues