RaceFlightOne/RaceFlightOne #245: FW420RC2: Configurator Telemetry Pane behaves erroneously

fujin, 2017-12-18:
After upgrading to FW420RC2 I noticed that the Telemetry pane behaves strangely:

Steps to reproduce:
* Upgrade FC FW to 420RC2
* Run through Wizard 1 to configure accelerometer (quad on flat surface, quad on nose)
* Switch to telemetry panel
* Rolling to right rolls the 3d model to the right
* Rolling to left rolls the 3d model to the right
* Pitching forward pitches the 3d model backward until it reaches 90deg, then it resets sharply
* Pitching backward pitches the 3d model backward until it reaches 90deg, then it resets sharply
* Yaws semi-constantly

Expected:
* Telemetry panel works normally after Wizard 1

PGIII, 2017-12-18:
Its likely because the acc needs to be recalibrated after the oreintation is set when you unplug and plug back in does it work normally ?

fujin, 2017-12-18:
@PGIII Still same behavior after re-plug.

SkeemoFPV, 2017-12-27:
Seeing the same issue here.

PGIII, 2018-03-07:
fixed in 500

closed 2018-03-07 by PGIII