RaceFlightOne/RaceFlightOne #256: TBS VTx Will Not Transmit When Armed if Battery is Plugged In On Profiles 2 or 3

8BBB, 2018-02-13:
Question: V500.147/Spark/Millivolt/TBS Race setup on two Big Deal 3" and one 5" Twig. All fly excellent and do everything well....except one odd thing. In order to turn the TBS video Tx on, they must be in Profile 1 when plugged in. After turning the VTx on, then they may be switched to Profile 2 or 3 and the VTx remains on. If the board is in Profile 2 or 3 when plugged in, the VTx will not allow it to be turned to on and transmit. This is the same on all three quads. No a big deal but I cannot find why this occurs. Any thoughts?  Also, this is with Taranis X7S/RXSR setup.  When running the RF LUA telemetry script, it must be on Profile 1 to recognize the VTx.  Thanks, Bob

PGIII, 2018-02-26:
we could check, do you have it set to go into pit mode on startup ?

8BBB, 2018-02-26:
Correct.  The Unify is in Pit Mode and begins transmitting normally VTXON with the same switch as the ARM switch.  Profile One must be selected during power up or it will not exit Pit Mode.

proggod, 2018-03-19:
This should be fixed, let me know if it is not.

closed 2018-03-19 by proggod

8BBB, 2018-03-20:
It works correctly in all three profiles now.  Thank you.