RaceFlightOne/RaceFlightOne #257: Motor wizard sets the wrong motor direction after flashing with BLheli32.

Derffpv, 2018-02-14:
Using a revolt 1 with bf 32 bit escs(wraiths).
After flashing beta flight 3.2 to my rf1 board and changing my motor direction correctly in Blheli32 suite(not the beta one), I flashed back to RF and ran all the wizards over again fine until the motor direction wizard. 
I confirmed my motor directions were correct before running the wizard but each time I do it it reverses all the motors again, or makes them all spin the same direction. As the wizard runs it says its setting the escs to dshot or entering Dshot mode I forget(i set them to multi on blheli) and reverses the direction of motors. Any reverse motor either stays that way or spins normal and any motor set to normal spins reverse.

PGIII, 2018-02-26:
You need to make sure your escs are on 32.31, the wizard sets the directions to what they were set to last to make sure the directions are as the wizard expects

Derffpv, 2018-02-26:
Alright I'll see which Blheli 32 version they are on when I get home
tonight. The second time I set the quad up I changed the directions again
in Blheli32 Suite and they were all spinning the right direction but when I
went to fly just flipped over. I'm just waiting on Crown City fpv to get
their shipment of bolts and revolts anywho. Thanks preston!

proggod, 2018-03-19:
Make sure you are using the latest version, and check the latest version and let us know if you still have issues v571

closed 2018-03-19 by proggod

Derffpv, 2018-03-19:
Got it all working great! Thanks for the help