RaceFlightOne/RaceFlightOne #262: FC Reboot w/ VTX Channel

miLstipple, 2018-03-19:
If you program VTX channel to same switch and position as VTX on the FC will reboot when you try and arm. Not all the time. Sometimes more than once. The behavior stopped when i removed VTX Channel from the switch.

PGIII, 2018-03-19:
VTXCHANNEL will take the vtx out of pit mode when it runs, also you dont need to set up vtx on to the arm switch since by default if its not setup then arming will take the vtx out of pit mode, vtx channel is best to have on something other than the arming position as well, opposite of arming is fine if you want it to always set that channel, i can add a check to vtxon to keep it from rebooting however

closed 2018-03-19 by PGIII

miLstipple, 2018-03-19:
Ur mistaken on one thing. I don't put VTX Channel or VTXon, on the arm
switch at all. Never did.