RaceFlightOne/RaceFlightOne #264: Bolt V2 + Millivolt OSD

MReFPV, 2018-03-20:
Hi, 

Bolt V2 Connected to Millivolt OSD FC (Raceflight 571) does not show correct Current Reading.

14 Ampere with no Lipo connected, 70A with Lipo Connected, but does not change on arming the Kwad.
Millivolt is set to BLheliS with Current Sensor (Solder Jumper).

[image]

[image]

[image]
[image]

Thanks

PGIII, 2018-03-20:
are you using 6s ? is correct on 4s ?

MReFPV, 2018-03-20:
4s is also not working, 11A with no lipo connected and 51A with lipo
connected.

51A (4s) 70A (6s) both values does not Change armed or disarmed Shows same
values.

MReFPV, 2018-03-23:
Tested Latest Flight One, still not working! Instead of current reading i
get a timer, if the Kwad is armed, it begins to count. Just Tested
betaflight on millivolt osd and guess what? It works! Too Bad raceflight
Hardware does not working with the Software it was made for 👎

siXsiXOneFPV <markusreit@googlemail.com> schrieb am Di., 20. März 2018,
22:43:

fsy78, 2018-03-23:
Same here!!! Very disappointing!  Thought such things were tested before selling.

PGIII, 2018-03-23:
it looks like the diagram for the current jumper may be backwards on the site, try reversing it, ive tested on my millivolt osd and everything is working, i found a bug that was causing current not to display properly, 577 will have the bug fixed and should be on the site shortly

closed 2018-03-23 by PGIII