RaceFlightOne/RaceFlightOne #270: vTx Profiles

trdaz, 2018-04-19:
I was thinking that it would be a good feature to have vTx profiles (similar to PID profiles) to pre set 3 (3 position switch) or more vtx band/channel/power settings so when at a race or just flying with a large group you can pre configure what the race is using and switch between them at the flick of a switch. For example: 
Position 1: R/1/200mw
Position 2: A/7/200mw
Position 3: R/7/200mw
Etc.

Also, Torben Frost suggested adding all of the power level options that the Unify and Tramp have available.

I think this would help a lot of people that don't run Taranis and even make them faster as well.

PGIII, 2018-09-10:
Currently there’s a vtx channel switch that lets you set a channel, can probably add some more at some point in the future

trdaz, 2018-09-10:
Hello,

Thanks for your reply.
I was thinking that on a 3 position switch one position would be open to
whatever you manually set and the other 2 positions could be preset to 2
other channels.
That way you can quick change in a race.  It might be useless with OSD and
LUA but was just an idea.

Thanks!

PGIII, 2018-11-29:
Current alpha will have a new mode that allows to cycle through 5 different preset channels

closed 2018-11-29 by PGIII