RaceFlightOne/RaceFlightOne #276: Config 594 firmware 590 acro level issue

michael4278, 2018-05-03:
Quad flys just fine. Turn on acro level and takes off. Have no control over throttle. Idles at 75% or so . Turn back full acro and can hover just fine.

PGIII, 2018-05-08:
try 595 and, also type dump and post that here after you upgrade

DJBlackhawk, 2018-09-17:
Michael4278 is this an issue for you when running the latest updates and or Beta 1.1.190?

DJBlackhawk, 2018-11-29:
> try 595 and, also type dump and post that here after you upgrade

This has been partially fixed with a combination of fixes in firmware and configurators 1.1x, 1.2xx, 1.2.75 and 1.2.76 it will only happen now under specific PID TUNE 2.5in Level/AcroLevel Pids still changes to SLD1=0.300 instead of 0.0290 to duplicate this save 2.5in tune to Profile 1, 2 or 3. Does not exist in any other tune I checked as of 1.2.76.