RaceFlightOne/RaceFlightOne #277: Smartaudio not working

carloscampos775, 2018-05-05:
SmartAudio with TBS Unify Pro 5G8 HV doesn’t consistently switch or read VTX channels in the OSD. VTX will also not consistently switch from pit mode to broadcasting when using the VTXON switch. Using OSD board with FW595, TBS Crossfire connected to TX1 and TX3 with Crossfire telemetry enabled, SmartAudio connected to RX3. Problem is consistent across multiple quads with identical hardware.

PGIII, 2018-05-08:
have you tried disabling crossfire telemetry, also how do you have your vtx placed in your quad and the wires ran, vtx control is working fine for me on multiple quads with both crossfire and spektrum, have you tried using tx3 also

carloscampos775, 2018-05-08:
SmartAudio works fine if I connect it directly to the Crossfire RX CH4.
Does SmartAudio not work with Crossfire telemetry enabled on TX3? What
firmware and RF board are you using. One build the VTX is on the top plate
and other build it's under the FC.

PGIII, 2018-05-09:
Under the Fc? Do you have a picture, and shouldn’t cause an issue but I don’t usually run crossfire telemetry(it’s banned at all the races) so it’s worth a shot turning off. My smartaudio has worked on every build I’ve used, all 500-595 builds I’ve tested smart audio has worked fine for me, used it last weekend on 595 no issue, with both the vtxchannel mode and changing through spektrum telemetry and the osd, so. It sure what differs from my builds to yours that makes it not work

DJBlackhawk, 2018-09-18:
Carloscampos was your issue resolved? Are you able to control TBS Unify Pro HV 5.8 using OSD after disconnecting Crossfire Ch4 Telemtry? If not have you tried the latest Beta 1.1.190?

carloscampos775, 2018-09-19:
I no longer fly Fightone. 

 

This issue can be closed

error414, 2018-10-06:
I have problem with smartaudio too, how to set band and channel is really unclear, I use TBS unifi. 

Look at the video, GUI is not able to set band and channel properly or I don't understand what it should do.  
And second thing, I had to set 500mW output power for 200mW real out power, If I set 200mW over lua script VTX was set to 25mW, Even my vtx desnt have 500mW. 

https://youtu.be/xFnm_rxLOuI

PGIII, 2018-11-29:
Significant changes in 1.2 to smart audio that should make even the nanos work

closed 2018-11-29 by PGIII