RaceFlightOne/RaceFlightOne #278: Acrolevel issues

tromax14, 2018-05-10:
Hi.
I try to find a reponse on discord but I haven't.

I make a video from the issues : 
https://www.youtube.com/watch?v=gXwbgUinYvM

The quad was very stable in levelmode but when am switching in acrolevel it start shake.
I have the problem with firmware 575 so i change for 595 but same.

I found issues about this problem on github:
https://github.com/FlightOne/RaceFlightOne/issues/269

and I try with this setting : (with FW 595 )
set slp1=2.500
set sli1=0.050
set sla1=40.000
set sld1=0.014
save

but it same.

Hope you can help me !

Regards

tromax14, 2018-05-11:
This setting don't help me (sorry) !
My dump without this modification : 

`#vr NAME:Flight One;VERSION:0.595.0.163 RC20;CONFIG:163
#fc HARDWARE:Flight One Revolt
#fl size=1799424
#fl total=16777216
#me modes active: 0
modes ARMED=8=70=100
modes LEVEL=6=-100=-70
modes ACROLEVEL=6=-29=30
modes FAILSAFE=0=0=0
modes LOGGING=0=0=0
modes BUZZER=7=70=100
modes LEDMODE=9=-29=30
modes LEDCOLOR=5=-29=30
modes DIRECT=0=0=0
modes VTXON=0=0=0
modes BRAINDRAIN=0=0=0
modes PROFILE2=0=0=0
modes PROFILE3=0=0=0
modes QUOPA=10=-29=30
modes PRELOAD=0=0=0
modes MOTORBEEP=7=70=100
modes AUTOKI=0=0=0
modes KILLSWITCH=0=0=0
modes VTXCHANNEL=0=0=0
modes OSDMENU=0=0=0
throttlecurve1 0=12=25=37=50=62=75=87=100
tpakp1 100=100=100=95=95=95=96=100=100
tpaki1 104=100=95=95=95=95=95=100=104
tpakd1 100=100=100=95=95=95=96=100=100
throttlecurve2 0=12=25=37=50=62=75=87=100
tpakp2 100=100=100=95=95=95=96=100=100
tpaki2 104=100=95=95=95=95=95=100=104
tpakd2 100=100=100=95=95=95=96=100=100
throttlecurve3 0=12=25=37=50=62=75=87=100
tpakp3 100=100=100=95=95=95=96=100=100
tpaki3 104=100=95=95=95=95=95=100=104
tpakd3 100=100=100=95=95=95=96=100=100
set mixer_type=0
set famx=100
set bounce_guard=0.000
set mixer_style=0
set esc_protocol=0
set m_dir=0
set esc_frequency=32000
set idle_percent=5.000
set idle_percent_inv=8.000
set spin_rec_str=750.000
set quopa_style=4
set esc_type=0
set mout1=3
set mout2=0
set mout3=1
set mout4=2
set mout5=4
set mout6=5
set mout7=6
set mout8=7
set led_count=8
set led_red=255
set led_green=0
set led_blue=0
set led_mode=6
set led_with_usb=0
set telem_smartaudio=0
set telem_sport=14
set telem_spek=0
set telem_msp=0
[... 232 lines cut ...]
set throttle_maxrc=1811
set aux1_maxrc=1811
set aux2_maxrc=1811
set aux3_maxrc=1811
set aux4_maxrc=1811
set pitch_map=2
set roll_map=1
set yaw_map=3
set throttle_map=0
set aux1_map=4
set aux2_map=5
set aux3_map=6
set aux4_map=7
set aux5_map=8
set aux6_map=9
set aux7_map=10
set aux8_map=11
set aux9_map=12
set aux10_map=13
set aux11_map=14
set aux12_map=15
set rc_calibrated=1
set bind=0
set short_throw=1
`

PGIII, 2018-05-23:
will have to look into it, closing this issue since this is already one open about acrolevel

closed 2018-05-23 by PGIII

tromax14, 2018-05-23:
Hope you find a solution !