RaceFlightOne/RaceFlightOne #289: CRSF Telemetry Broken on builds after 1.1.140

Rafyel, 2018-07-27:
When upgrading to any build after .140, telemetry is broken; .141, .150, .152, .157, etc. If you then DFU flash back to 140 and apply the automatic backup of your settings that’s prompted after the flash, it breaks 140. Even you a manual backup of your settings also then it’ll break .140. So it seems a setting somewhere is getting changed during an upgrade in all the builds after 140 that also gets backed up for restoring your settings of the last Revolt FC flashed by the Configurator that’s possibly pertaining to the telemetry for CRSF. The only way to keep telemetry working with CrossFire is to do a clean install of 1.1.140 without any previous settings.

PGIII, 2018-08-02:
can you post dump from the version that works and one from the one that doesn't

Rafyel, 2018-08-02:
Files attached.

Rafyel, 2018-08-02:
[1.1.140.184 - BlHeli_32 dShot1200 (CRSF Broken Voltage & mAh Telemetry).pdf](https://github.com/FlightOne/FlightOne/files/2252582/1.1.140.184.-.BlHeli_32.dShot1200.CRSF.Broken.Voltage.mAh.Telemetry.pdf)
[1.1.140.184 - BlHeli_32 MultiShot (CRSF Working Voltage & mAh Telemetry).pdf](https://github.com/FlightOne/FlightOne/files/2252583/1.1.140.184.-.BlHeli_32.MultiShot.CRSF.Working.Voltage.mAh.Telemetry.pdf)
[1.1.157.184 - BlHeli_32 dShot1200 (CRSF Broken Voltage & mAh Telemetry).pdf](https://github.com/FlightOne/FlightOne/files/2252584/1.1.157.184.-.BlHeli_32.dShot1200.CRSF.Broken.Voltage.mAh.Telemetry.pdf)
[1.1.157.184 - BlHeli_32 MultiShot (CRSF Broken Voltage & mAh Telemetry).pdf](https://github.com/FlightOne/FlightOne/files/2252585/1.1.157.184.-.BlHeli_32.MultiShot.CRSF.Broken.Voltage.mAh.Telemetry.pdf)

Rafyel, 2018-08-02:
The previous files from 1.1.157.184 were an in-place upgrade from 1.1.140.184. These files are from a clean DFU to 1.1.157.184. In these I noticed that the initial voltage when plugging in the battery is audible with telemetry and displays in the OSD, but doesn't change ever.

[DFU - 1.1.157.184 - BlHeli_32 dShot1200 (CRSF Partially Broken Voltage & mAh Telemetry).pdf](https://github.com/FlightOne/FlightOne/files/2252748/DFU.-.1.1.157.184.-.BlHeli_32.dShot1200.CRSF.Partially.Broken.Voltage.mAh.Telemetry.pdf)
[DFU - 1.1.157.184 - BlHeli_32 MultiShot (CRSF Partially Broken Voltage & mAh Telemetry).pdf](https://github.com/FlightOne/FlightOne/files/2252749/DFU.-.1.1.157.184.-.BlHeli_32.MultiShot.CRSF.Partially.Broken.Voltage.mAh.Telemetry.pdf)

PGIII, 2018-09-11:
do you still have this issue on 1.1.180?

Rafyel, 2018-09-11:
It seems to work with MultiShot, I’m not sure about its accuracy, but it doesn’t work on dShot. The OSD is definitely off calibrated when using dShot1200, but appeared to be okay with MultiShot.

closed 2018-09-13 by PGIII