RaceFlightOne/RaceFlightOne #291: Oscillations since 1.1

Rouvster, 2018-08-03:
There are several people having issues with weird oscillations in the mid/full throttle range that appeared since FW 1.1.

Some developers are aware of the issue and are trying to fix it but it would be nice to have a heads up on here to see where this issue goes.

Apparently it isn't PID or filter tuning but an issue with the code, problem persist on 1.2.2

Cheers

PGIII, 2018-08-03:
Sounds like a tuning issue, the default tune changed in 1.1, you can always use your same tune, for the filters the older versions used two frequency’s on gyro first at 120 and second and 70, and a biquad On d at 80 or whatever you have kd filtering set to in the old version

Rouvster, 2018-08-03:
I was talking with ProgGod about it and he states it’s not a tuning issue (tried presets blackbird, Skitzo and default), all do the same.
logfile is on discord channel test monkeys, I could find it for you.
Preston is also aware of this issue, several people mentioned it on Facebook.

The same tune runs perfectly smooth on 0.500, and 1.1, 1.2 isn’t. 
Flashed t 1.2.4 am going to try it this evening or tomorrow

PGIII, 2018-08-03:
Do you have a dump from each version if your filters aren’t set right it won’t be the same

DJBlackhawk, 2018-09-13:
Rouvester, out of curiosity is this on a RevoltV2, V3 or RevoltOSD? Instead of Alpha build, have you tried the latest Beta 1.1.190 using Default PIDs and not touching the Filters, or Rates? Also, using Configurator 1.1.178 to do a DFU Flash. Also, are you running any other peripherals on other UARTS like Telemetry or Smart Audio or CRSFR? Just wondering as I have not noticed this issue on my builds but I haven't installed a RevoltOSD and CRSFR on a same build yet ...

DJBlackhawk, 2018-11-16:
> Do you have a dump from each version if your filters aren’t set right it won’t be the same

This user hasn't responded, hasn't provided dump, it was his first bug post, no other correspondence. No one else has reported similar issues, I can't duplicate on any of the newest Betas.

closed 2018-11-29 by PGIII