RaceFlightOne/RaceFlightOne #299: When Arming video goes to snow, Tramp telemetry

2dogrc, 2018-09-06:
on 175, when you arm with Tramp Telemetry the video goes to snow for a few seconds. Telemetry wire TX3.  This was not really an issue before, but without double arm, when you arm after entering turtle mode, you can not see if the quad flipped over.
Using iFlight iForce V2 on Revolt, and Revolt OSD.

PGIII, 2018-09-10:
So you get snow when you arm ? Does it go into pit mode when disarmed or you just lose video when arming?

2dogrc, 2018-09-10:
When I disarm it goes to snow as well.  Then video comes back.  When I enter the OSD to set my channel it shows A 1, but my channel is Race 1.  I can change it the channel and power okay in the OSD, but when I open the OSD again, it shows A 1.

Here is a video of the arm/snow sequence.  When you see snow that is me arming.
https://www.youtube.com/watch?v=hmiyFJr0lE8

DJBlackhawk, 2018-09-13:
2dogrc have you tested this issue with Beta build 1.1.190?

When you set up your quads do you go into Configuration Tab and Setup PitMode Type? (Just asking Cause I normally don't unless I'm testing it or building a racequad)

In the Configuration Tab running the VTX PitMode Type Wizard. 

Also, what VTX Band and VTX Channel do you have set as default? I always set these to my favorite Band and Channel and hit save.

If you un soldier Tramp Telemetry does the problem go away (I would hope so, but if for some crazy reason it still does I would be totally lost as I don't have a Tramp to test anymore).

2dogrc, 2018-10-18:
Test  on 1.246..still happens.

Have not tried messing with wires...flashed board to BF, no static when arming.

2dogrc, 2018-11-14:
Fixed on later versions, 1.2.5 and up I believe, for sure fixed on 1.2.64.

closed 2018-11-14 by 2dogrc