RaceFlightOne/RaceFlightOne #300: Proshot is DANGEROUS!  Motors spin up uncommaned when disarmed

2dogrc, 2018-09-06:
When selecting Proshot, any of them, on BL Heli32, some motors twitch and even spin up when quad is disarmed.  I recommend removing these options until they are 100%, its just dangerous.

DJBlackhawk, 2018-09-06:
Which version are you on? 
Does it happen in Level or Acro mode (no mode selected) or only AcroLevel?

2dogrc, 2018-09-07:
Tha was on 1.29and 1.158 I believe. It happened when the quad was not armed and sitting on the table going through the wizards. 

Mark Grohe
2dogrc@gmail.com
682.552.6494
Follow us on Twitter: www.twitter.com/2dogrc
Our YouTube Channel
www.2dogrc.com/2dogrc

DJBlackhawk, 2018-09-10:
Is this still occurring after beta update 1.1.176?

2dogrc, 2018-09-10:
I have not had a chance to try it.

PGIII, 2018-09-10:
The latest version of Blheli32, 32.6 I believe, is supposed to fix the issue with proshot, I’ve seen similar reports of motors twitching on betaflight as well with proshot and blheli32, i plan on testing the new version this week to see if it does fix the issue

DJBlackhawk, 2018-11-16:
I noticed subtle twitch and slight motor wind ups but very brief and barely a full revolution with 32bit ESC's on 32.4. I haven't tested Blheli32 firmware 32.6 because PGIII mentioned that 32.6 was still in beta an could impose other bugs or issues?

2dogrc, 2018-11-16:
Tested on 32.6, does not spin up.

closed 2018-11-16 by 2dogrc