RaceFlightOne/RaceFlightOne #301: A mode to turn off the OSD Display.

2dogrc, 2018-09-10:
I wonder if it would be possible to assign a mode to turn off the OSD on the screen.  When I race, Iike a blank screen, but I also like to do a check of my battery voltage and make sure its my quad in the goggles by looking for the Pilot Name variable I have set at the bottom of the screen.

PGIII, 2018-09-11:
i can see about adding this, you just want it to make the screen completely blank ?

DJBlackhawk, 2018-09-13:
I don't think he wants the screen to go completely blank, I think he just wants the OSD overlay to completely go away, like a mode to switch OSD Off.

2dogrc, 2018-09-13:
Just the OSD info disappear so you have a clean picture. 

Mark Grohe
2dogrc@gmail.com
682.552.6494
Follow us on Twitter: www.twitter.com/2dogrc
Our YouTube Channel
www.2dogrc.com/2dogrc

PGIII, 2018-11-06:
Just added this in 1.2.60, currently in Test Monkeys on discord

closed 2018-11-06 by PGIII

2dogrc, 2018-11-07:
I saw that!  Thank you very much.

Mark Grohe
support@2dogrc.com
www.2dogrc.com
800-517-3810 ext 1
Twitter: http://www.twitter.com/2dogrc
Facebook: 2dogrc
<https://www.facebook.com/pages/2dogrcs-RC-Warehouse/155034941272913>
YouTube:2dogrc <http://www.youtube.com/2dogrc>