RaceFlightOne/RaceFlightOne #306: MPS telemetry bug

error414, 2018-10-09:
Hello,

In MSP telemetry is bug what cause impossibility to parse all frames properly. In frame MSP_STATUS missing last byte (CRC).  

https://github.com/FlightOne/FlightOneOpenSource/blob/master/src/flight_controller/src/telemetry/mspTelemetry.c#L108  packed size is not 16 but 17!

PGIII, 2018-11-29:
What is it you’re using MSP for ? I’m not sure its implemented fully as its not really a feature used/needed by many

error414, 2018-11-29:
I use MSP for MWOSD, Flightone OSD has not implemented all functions what I like. Are you going to remove MSP from flightone? Bug, I mentioned is easy to fix, it is just broken frame, you send wrong packet size, just replace 16 for 17.

artemaminov, 2020-03-03:
I think my issue is related to this. MWOSD not receiving RC data, so I can't get in to OSD menu with sticks and there is no RSSI data also.
Millivolt v1 using