RaceFlightOne/RaceFlightOne #312: OSD in Revolt OSD is disapearing if video standard is switched while powering on.

Mmarsian, 2018-11-04:
The problem occures because I'm on PAL and NTSC :-(
The Runcam Eagle 2 Pro (recommended as part of my Skitzo Nova build) is switchable between NTSC and PAL. Since my gear and FPV monitor requires PAL I've set my cam to PAL. But despite this setting the cam starts with NTSC when I connect the Lipo. Then after a second (or so) it reads the settings and switches to PAL. During this second I can see the OSD but then the OSD is disapearing along with the video standard changing to PAL. My FL1 settings are on internal OSD and PAL, but that really doesnt matter. FL1 maybe looks for the video standard just once while startup. It discovers a NTSC signal, so it arranges for NTSC. But then the cam switches to PAL and that makes the OSD disapear. I reported this problem when I started with FL1 with the 1.1.140 beta version. Shortly after that they fixed that in a later version and now I find this reintroduced in the current beta.

Mmarsian, 2018-11-05:
Please I need a quick workaround for this problem. I noticed that if I change the configuration and the FC reboots completely with the camera being still in PAL, I can make the OSD appear. So out in the field I don't have the computer with me to make an USB connection to the FC, how I can force a reboot of the RevoltOSD without disconnecting the Lipo, any stick command?

PGIII, 2018-11-06:
1.2.60 should fix this, its posted in test monkeys on discord

closed 2018-11-06 by PGIII

Mmarsian, 2018-11-06:
Thanks for the quick answer but:

Sorry, just tried it, DFU flash, no import, first wizard (Flightcontroller) hangs in the stage where I put the quad on its nose. After a while: "Calibration faild, make sure ...." After that I have to cut the connection (power) and restart. Did that several times without success.
Do I need a new configurator? I've got 1.2.50.