RaceFlightOne/RaceFlightOne #317: Fw 1.2.46 config 1.2.50 issue using flysky Nirvana

antoni9, 2018-11-13:
Quad won't arm after a clean install. Plugging into computer shows the radio isn't detected, however running through the setup process is possible and radio showing, but only while it's plugged in. Can't use it standalone, it's like the changes aren't being saved..

PGIII, 2018-11-15:
What do you mean by plugged in what’s plugged in the battery ? And what pad do you have the ibus wire soldered to and are you having Dshot ?

antoni9, 2018-11-15:
MillivoltOSD, Spark32 [Multishot]
iBus goes to Rx3

I meant that the radio channels are detected after setting up the radio, but when disconnecting the FC from computer - the quad doesn't arm. Connecting to computer again shows the same thing - no radio detected.

DJBlackhawk, 2018-11-15:
I us must be connected to TX1 https://wiki.flightone.com/2017/08/03/installing-fs-a8s-receiver-for-ibus/

antoni9, 2018-11-15:
OK, but why does it work on non-beta fw 1.1x ?

antoni9, 2018-11-15:
And it also works while the quad is connected to configurator..

DJBlackhawk, 2018-11-16:
antoni9 why doesn't what work on non-beta fw 1.1x? RX3 or your iBus Receiver? Are you saying your setup works on non-beta versions but on 1.2.46 receiver does not work? There were some arming issues changes you should really upgrade to Configurator 1.2.62 and DFU flash to firmware 1.2.62 do NOT import settings. Disconnect from the computer, plug in lipo and use a multimeter to check that iBus Receiver is getting appropriate voltage. Sounds like the USB is powering the FC and Receiver but when you switch to Lipo your Receiver is not getting the correct voltage and turning on. I will reach out on Discord so you can post a picture.

PGIII, 2018-11-29:
Closing this as its a setup related issue, if you are still having problems you can contact us on discord, flightone.com/discord or message the Flightone page on facebook

closed 2018-11-29 by PGIII