RaceFlightOne/RaceFlightOne #327: No RC data in MSP package

artemaminov, 2020-03-03:
So I can't get in to MWOSD menu with sticks and there is no RSSI data also.
Millivolt v1 using

SkeemoFPV, 2020-03-03:
Are you using FlightOne or FalcoX ?

artemaminov, 2020-03-04:
FlightOne 1.2.94.193
Flight One Revolt (Millivolt v1)
MWOSD 1.9.1.4 FW

SkeemoFPV, 2020-03-04:
Unfortunately what you’re trying to do won’t work, you need to flash a special fw to the OSD board. You will need to join the FlightOne Discord channel if you haven’t already, then look in External OSD topic. Pinned in that section is the FW you need and a set of instructions explaining what you need to do.
https://flightone.com/discord.php

artemaminov, 2020-03-04:
Flashed ttyosd.
It works, but some characters are corrupted % -> &
[image]

closed 2020-03-06 by artemaminov